mit-racecar/racecar #1: Move steering angle clipping further upstream

Winter-Guerra, 2017-02-28:
According to the VESC config, the radian angles max out around +−0.264. Currently, clipping downstream in the VESC makes Gazebo sad. The clipping should move upstream in the node architecture.

mboulet, 2017-02-28:
Gazebo can't handle a command of 0.264? Or is the problem that gazebo does not use the vesc node?

Winter-Guerra, 2017-02-28:
The VESC clips steering at around that value to match hardware. However, Gazebo does not know about this limitation and will try to turn wheels in an angle greater than that if commanded to. This breaks the physics in gazebo and causes the car to move sideways.

mboulet, 2017-02-28:
Then it seems like the problem is that the simulator is not a good simulation of the actual hardware.