mit-racecar/racecar #5: How to navigate with RaceCa

NevzatBOL, 2018-04-20:
I want to make real time navigation with racecar. When I run the navigation packages, it sends me the data of type cmd_vel Twist. How can I check the motors using this data?

CapitalZe, 2018-05-07:
Do you mean the navigation_stack or a navigation package in this repository?

NevzatBOL, 2018-05-07:
I didn't find navigation package this repository. I created a navigation_stack in this repository. I do not know how to use a transformation to control engines.

CapitalZe, 2018-05-07:
I'm having a similar issue. I also tried nav2d, before I found the
Formula1Epoch repository on github. I'll be giving that a go tomorrow.
Let me know if you make any progress, i'll be curious to learn how to do it.

Cheers!
[image: 5]

xoxota99, 2018-06-07:
What is the data in the Twist message that you're getting? Could it just be an error value for steering (and assume constant velocity to the motor)?

closed 2019-01-27 by NevzatBOL