mit-racecar/racecar #7: How to train the car on new course

ArtlyStyles, 2018-05-26:
Hi, is there any documents about how to collect training sample and train the car for a new course?

ArtlyStyles, 2018-05-26:
Can anyone help?

xoxota99, 2018-06-07:
I would also love to know the answer to this. At a high level, as I understand it, you launch the joystick teleop, capture all lidar data to a rosbag or log file, then use the captured to create a map offline with some MATLAB magic. Repeat that a couple of times to create an "averaged" view of your track, and export to PNG, I guess.

mboulet, 2018-06-29:
Hi, this repo does not currently contain the "high level" autonomy that you describe. The intent of this repo is to provide a platform with low-level control for others to build such autonomous capability on their own. But it sounds like a great idea - you should try it!

closed 2018-06-29 by mboulet