mit-racecar/racecar #11: Need to center the steering

ArtlyStyles, 2018-08-26:
I built a Racecar J and could use teleop to drive it. I found, however, my car always turn right a little bit, even when I drive it straight. I think the steering is not centered. Is there anything I can tune with the BLDC tool? Or I have the tune the car mechanical parts?

mboulet, 2018-08-26:
I'm not an RC car expert, but I believe that there is a way to adjust the steering mechanically. But first, I would suggest tuning the steering_angle_to_servo_offset parameter in vesc.yaml if you are using the throttle_interpolator.py node.

closed 2018-08-26 by mboulet

ArtlyStyles, 2018-08-27:
hi @mboulet Chaning the steering_angle_to_servo_offset did fix the issue when the car was moving forward. However, when the car was moving backward, it still turned right slightly. Do you know how to fix that?

Thank you very much.