mit-racecar/racecar #12: Understanding ackermann_cmd_mux

subodh-malgonde, 2018-09-21:
Is there any documentation for the `ackermann_cmd_mux` package? When I fire up the Gazebo simulation I can see a complex graph (rqt_graph). I understand that this package reads inputs from different sources (joystick, safety controller and autonomous algorithms like the wall follower) and decide which one to use based on the joystick inputs and the priorities. The priorities for the topics are in the following order:

1) /vesc/ackermann_cmd_mux/input/teleop (highest priority)
2) /vesc/ackermann_cmd_mux/input/safety
3) /vesc/ackermann_cmd_mux/input/navigation (lowest priority)

My understanding is based on this presentation - [Racecar Basics](https://lchsracecar.weebly.com/uploads/5/4/6/4/54642095/7_racecarbasics.pdf)

### My question

If I have to use an autonomous algorithm along with using a joystick then to which topic should the autonomous algorithm publish control commands to? Currently the joystick is publishing to `/vesc/low_level/ackermann_cmd_mux/input/teleop`. I have tried `/vesc/ackermann_cmd_mux/input/navigation` and `/vesc/high_level/ackermann_cmd_mux/input/nav_0`, none worked.

### Background 
I am right now testing a "wall following algorithm" in the F1/10 simulator. My node reads the laser scans, computes the PID control values to make the car go parallel to the wall and publishes a AckermannDriveStamped message on the topic `/vesc/ackermann_cmd_mux/input/teleop`. It works fine. 

Now I want to be able to use the joystick, just in case if I have to course correct, while my wall following algorithm is running. 

So I understand that my wall follower node cannot use the /vesc/ackermann_cmd_mux/input/teleop topic anymore since that would mean that the algorithm commands would have the highest priority. So I change the wall follower's topic to `/vesc/ackermann_cmd_mux/input/navigation`. I fire up the joy node and I observe that I am only able to control the car with the joystick. When I switch over to the autonomous operation (by holding down right bumper **RB** button), I observe that the car is not controlled.

Then I change the topic to `/vesc/high_level/ackermann_cmd_mux/input/nav_0` (based on sample solutions posted by MIT Racecar course TAs). Still nothing happens.

subodh-malgonde, 2018-09-22:
I figured out the problem. The topcis were setup fine. I was using the Joystick incorrectly.

If you are using the Logitech Gamepad F710 then the way it is setup by default is:

1. teleop requires LB button to be active 
2. Autonomous mode requires RB button to be active if the joystick is plugged in. Otherwise it does not require any button to be active

The mistake I was - I was holding the RB and LB buttons down simultaneously. I was under the impression the dead man's switch (LB button) worked in both cases - teleoperation as well as autonomous navigation. I was wrong. It only works for joystick operation.

[image]

And the topics are:

1. _joy_teleop_ node should publish to `/vesc/low_level/ackermann_cmd_mux/input/teleop`. In the source code it publishes to `low_level/ackermann_cmd_mux/input/teleop` you can add the prefix `/vesc` by remapping in the launch file.
2. The autonomous algorithm should publish to `/vesc/high_level/ackermann_cmd_mux/input/nav_0` (there are also other topics ending with `nav_x` where x is one of  `[1,2,3,default]`. The value of `x` determines the priority. See this file [high_level_mux.yaml](https://github.com/mit-racecar/racecar/blob/master/racecar/config/racecar-v2/high_level_mux.yaml) for the order of priorities.

closed 2018-09-22 by subodh-malgonde

ArtlyStyles, 2019-09-19:
Hi @subodh-malgonde. I have two questions:

1. What is the difference between /vesc/high_level/ackermann_cmd_mux/input/nav_0 and /vesc/high_level/ackermann_cmd_mux/input/navigate?

2. Where is the teleop node source code?

ArtlyStyles, 2019-09-19:
Another question is, if I want the car to stop immediately in a safety emergency situation, does the teleop node have a button I can press? If want, what is the best way to implement it?