mit-racecar/racecar #16: Car can not turn

ArtlyStyles, 2018-12-29:
I have a very strange problem I hope you can help me with. My MIT-Racecar can go forward and backward but it can not turn. Neither the teleop or ros command can make it turn. When I tried to turn the front wheels by hand, it feels very stiff but turntable. I am wondering what's going on and how can I debug. Is servo bad? Is there anyway to check with the BLDC tool?

mahmoodjanabi, 2018-12-30:
If I had to guess, I'd say you have uploaded the wrong firmware to the VESC. While it's not mentioned, you should always use the servoout.bin firmware 2.18 and not 3.xx. I have uploaded the proper firmware that you can download from my Github: https://github.com/mahmoodjanabi/Caroline/tree/master/VESC%20Firmware

Try that and let me know how it goes!

Mahmood Janabi

ArtlyStyles, 2018-12-30:
No, the car was fine yesterday. I did not update the software

luroess, 2020-01-14:
Which hardware version of the VESC are you using?