mit-racecar/racecar #18: Can had problem to start with LiPO battery

ArtlyStyles, 2019-05-11:
I used to use NiMH and everything works fine. To get more power, however, I replaced the NiMH battery with 4S LiPO.  After that, I found that my car had difficulty to start. Please see the video here https://www.youtube.com/watch?v=D1XGa-vANRY   I am not sure how to fix it and why I did not have the same problem with NiMH battery. Anyone uses LiPO battery and experienced the same problem?

EwingKang, 2019-05-13:
By pure guessing: I heard some clicking noises the moment you throttle your vehicle. Sounds to me the gearbox is having some gear slipping issue? Maybe Lipo is too powerful for your transmission.

ArtlyStyles, 2019-05-13:
Thanks. Someone told me I need to "gear down". But not sure what this means and how to do it...

EwingKang, 2019-05-15:
@ArtlyStyles I suppose that means you need to either use a gerbox with higher gear ratio, i.e. slower wheel turn with less motor rpm and hense the "gear down".
You can also try to use a stronger gearbox with matel gears, or simple use a less powerful battery, a 3 cell one maybe?